leodesigner/espNowFloodingMeshLibrary2 #5: upgrade

ozi2277, 2023-07-18:
Hi! Your library is amazing!! I admire you how much time you put into modifying this library. I've been using it for a while now and I have a question. Will there ever be an upgrade to be able to use the library for esp32 version 2.0 or higher ??

leodesigner, 2023-07-19:
Thank you @ozi2277  :)
Do you mean Arduino framework version for ESP32?
I had no chance to test it, what kind of compile errors you are getting ?

ArminPP, 2023-07-19:
Hello ozi2277,
I use this library in PlatformIO with version >2.0, I think.
And it works like a charm, thanks again @loedesigner :-)

PIO says after compiling:
 - framework-arduinoespressif32 @ 3.20009.0 (2.0.9) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5

So I think 2.0.9 is the Version ?

Five months ago I wrote a little test program for this library: https://github.com/ArminPP/M5Atom_ESP-MESH_Test
It is written for M5Stack Atom, but should also work also with a plain ESP32.

leodesigner, 2023-07-25:
I guess we can close the issue now.
Thanks.

closed 2023-07-25 by leodesigner