leodesigner/espNowFloodingMeshLibrary2 #6: Esp32-c3 compatibility ?

Zhairgling, 2023-07-27:
Hi, thank's for your work !

I'm working with your fork. That's working fine with several esp32 like devkit and weemos d1mini but I can't get it working with the esp32-c3 board. 
Any advice ?
Thank you again mate !

Zhairgling, 2023-07-27:
#define DISABLE_CRYPTING true

"solve" the problem

Zhairgling, 2023-10-04:
It is not an issue with espNow but with the crypto lib used

brickstuff, 2023-10-20:
So using a C3 board, all messages are sent in the clear? Is this due to the single-core processor or...? I was all set to try this using low cost C3 modules 😞 but need the encryption.

leodesigner, 2023-10-23:
Do you get any specific compilation errors?

314159-r, 2024-04-20:
I'm not having compilation errors related to this issue. 
For me the ESP32-C3 node device hangs. After sending time-sync message, and when receiving any message. So the first encryption works, but then subsequent decryption hangs...

I Googled, and found that the commands: `esp_aes_acquire_hardware()` and `esp_aes_release_hardware()` in the encrypt and decrypt functions in EspNowFloodingMesh.cpp, seems to be excessive. So I tested to comment them out - And now it works... :)

Comment or remove these lines in `EspNowFloodingMesh.cpp`:
[837](https://github.com/leodesigner/espNowFloodingMeshLibrary2/blob/47be15a4b6f730a91963ce0a22108eaeb0d0e54a/EspNowFloodingMesh.cpp#L837), [839](https://github.com/leodesigner/espNowFloodingMeshLibrary2/blob/47be15a4b6f730a91963ce0a22108eaeb0d0e54a/EspNowFloodingMesh.cpp#L839), [873](https://github.com/leodesigner/espNowFloodingMeshLibrary2/blob/47be15a4b6f730a91963ce0a22108eaeb0d0e54a/EspNowFloodingMesh.cpp#L873), [875](https://github.com/leodesigner/espNowFloodingMeshLibrary2/blob/47be15a4b6f730a91963ce0a22108eaeb0d0e54a/EspNowFloodingMesh.cpp#L875)

**!! I have only tested this fix on a ESP32-C3 device**, so I don't know how it works for other ESP32 variants.

That fixes the encryption problem - But... I can only use my ESP32-C3 node device together with an ESP32 master. And ESP8266 node devices only works with ESP8622 masters. I'll open a new issue for that - Tomorrow...