leodesigner/espNowFloodingMeshLibrary2 #10: Will ESP NOW Mesh Support 1000 nodes Connected in a line(Street Light) each 50-100 meter apart

mightChamp, 2024-05-29:


leodesigner, 2024-06-04:
The TTL limited to one byte, so probably you can reach up to 255 nodes.
But with modifications of TTL field it should be possible to reach over 1000 nodes.

See:

https://github.com/leodesigner/espNowFloodingMeshLibrary2/blob/47be15a4b6f730a91963ce0a22108eaeb0d0e54a/EspNowFloodingMesh.cpp#L66

mightChamp, 2024-06-06:
Thanks for response, I need to send data to specific node, but there is no function to do that, How Can I do this?

VladCiocan, 2024-07-05:
> The TTL limited to one byte, so probably you can reach up to 255 nodes. But with modifications of TTL field it should be possible to reach over 1000 nodes.
> 
> See:
> 
> https://github.com/leodesigner/espNowFloodingMeshLibrary2/blob/47be15a4b6f730a91963ce0a22108eaeb0d0e54a/EspNowFloodingMesh.cpp#L66

So if ttl would be 2 bytes(max 65k) could we go to 3-4k nodes? Do you think there are other limitations?

leodesigner, 2024-07-06:
In theory, this is the only limitation to my knowledge.
I think, it make sense to create a virtual test config without implementing it in hardware.

VladCiocan, 2024-07-06:
I'm currently trying to implement it on heltec wireless tracker cause I want the nodes to share sensor data plus gps position. But i'm having a lot of compatibility issues. Any suggestions?

leodesigner, 2024-07-06:
> I'm currently trying to implement it on heltec wireless tracker cause I want the nodes to share sensor data plus gps position. But i'm having a lot of compatibility issues. Any suggestions?

It's hard for me to guess, the code should be compatible with the new framework version and compile under platformio.org.

Any particular error you are getting?

VladCiocan, 2024-07-18:
I have a question regarding ttl. If we store last x no of messages so we can eliminate duplicates, why do we use ttl?

leodesigner, 2024-08-29:
ttl is just another way to restrict packet propagation. Sometimes if you are reaching a nearby device you don't need to propagate packet to distant nodes